neurogears/ranson-rig #29: sync pulses

ransona, 2021-02-22:
The sync pulses produced by bonvision will be acquired by a DAQ system we use called 'Timeline' at 1000Hz. Timeline also acquires pulses from the microscope and cameras etc, and will also acquire the photodiode signal. We will use the sync pulses from bonvision to know the time (in timeline's time base) when different bonvision/behavioural events happen (like rewards, stimulus onsets etc). 

The optimal thing would probably be that when the rig.experiment command is called (which starts a new experiment right?), bonvision starts sending out alternating high/low pulses at eg 1 Hz, and timestamps these in bonvision time  (or what ever the timescale that is stored in the experimental data file that shows when behavioural evenbts etc happen) in a log file for the experiment.

glopesdev, 2021-06-04:
This is related to #23 and has now been fixed. Logging of individual frame flips in screen time is also included for each trial.

glopesdev, 2021-06-13:
When synchronizing pulses across non-frame-locked displays it becomes hard to ensure accurate logging of individual frame flips in the case of a dropped frame (which might be dropped in one display, but not in another).

To prevent this, some alternate strategy of identifying frame presentation time is required, e.g. a fixed pseudo-random sequence of high/low swaps.