neurogears/ranson-rig #39: blips in timestamps of digital sync pulses

ransona, 2021-07-09:
Sometimes timestamps of digital pulses seem to be momentarily out of sync with an external timesource. The amount of desynchrony (in ms) is approximately a multiple of monitor refresh rate.

glopesdev, 2021-07-15:
Are these input or output pulses? Is it the photodiode data or some other signal? Can you point us to an example dataset in the shared folder where this is visible so we can better understand the context?

ransona, 2021-07-15:
From my email on this:

There are a few peculiarities which you can see in the plot below from 30 minute experiment: 2021-07-08_30_TEST. 
1) the 2 systems seem to have a slightly different notion of time, and linearly drift out of time at a rate of about 160ms per hour. (the linear trend you see below).
2) there are some blips where they momentarily get more desynchronised then recover (the bigger peaks below). I´m unsure whether the blips come from bonsai or our system incorrectly timestamping. 
[image]

The linear drift is expected (I think?) and not a major problem. The blips can also be dealt with post hoc but I´m wondering where they come from and if they can be prevented in the first place. I assume they are from bonsai as if you look at the histogram of the amplitude of the blips they are clustered at the monitor frame interval:

[image]

We can clearly smooth this out with a linear fit given that that the systems go back into time after blips but I´m wondering what this means for timestamping of ´intended´ trial onset times etc?

ransona, 2021-07-15:
In the first figure above remember that the NI DAQ starts ~9.5 secs after the Bonsai acquisition

glopesdev, 2021-07-15:
Thanks, I will have a look. The temporary blips could be frame drops, in which case the update frame logic is indeed able to recover. I'm surprised to see so many drops though, unless there was a screen monitoring process like VNC or team viewer on?

There is another option which is the update loop can indeed submit commands late. If this happens though, it is not always the case that there will be a frame drop, as there are buffers in the system to compensate for this.

An interesting test might be to use the frame number column in the CSV file and multiply that by 1/120 (update rate is 120Hz i believe). That should give a logical reconstruction of frame timing which might line up even better wiht the NI-DAQ signal.