neurogears/ranson-rig #42: does trial number 0 or 1 in frames file indicate first trial onset time?

ransona, 2021-07-09:
We use the trial number column in the Frames file to know the bonsai time of trial onsets. Does the first timestamp in the frames file indicate the time of 'intended' first stimulus onset, or is it when the new experiment was started (i.e. with rig.experiment(meta))? Or in other words is the trial number in the frames file column 3 zero on one based?

glopesdev, 2021-07-21:
It should be zero based, but is now not aligned to any particular stimulus onset, as we discussed before it might be easier to just have a continuous signal going out both the photodiode and Harp outputs.

This does mean however there is no trial information on the synchronization pulse, which might make it more difficult for certain types of analyses. This could be mitigated by logging the absolute frame timestamp when trials are initiated using the start trial command. Is that what you had in mind?

ransona, 2021-07-22:
I think we need to discuss this to make sure we get it right as this is really critical to getting a fully working version, and I'm struggling to explain/understand 100% clearly what's needed here. 

For a start: if you imagine our NI DAQ, which is storing the microscope frame pulses and timing pulses from bonvision - can you see what the method would be to know on our DAQs timescale when each trial was _intended_ to start (note: not when it did start, as I understand there can be a dropped frame)? i.e. if we had a grating which _should_ come on at t=0 secs and go off at t=1 secs, how do we know which microscope frames to look at?

I see the frame events file allows us to know in Bonvision time (correct?) when the sync pulse edges are, but I don't now see where we can find in the same timescale when each trial should have started... I thought this was in the 4th column of FrameEvents but from what you say above it seems not?

Anyway, I think best to have a quick chat? Any chance we could do that in the next few hrs?