neurogears/ranson-rig #43: wrong number of trial files

ransona, 2021-07-09:
Sometimes there seems to be one more trial file than expected. For example in this experiment:
\\AR-LAB-NAS1\DataServer\Remote_Repository\TEST\2021-07-09_05_TEST
I did 9 trials but there are 10 trial files (i.e. 0 - 9). The final one ('2021-07-09_05_TEST_00009.csv') does not contain trial parameters.
[image]

ransona, 2021-07-09:
I tested some more and it seems to be random whether it makes more or less trial parameter files.

glopesdev, 2021-07-22:
I think I understand now what is happening and there was actually a race condition preventing the proper creation and logging of trial metadata. Hopefully this will be fixed with the following commit.

commit 7d50d309 (2021-07-22): Avoid sampling to ensure no trial metadata is lost

Fixes #43
closed 2021-07-22 by glopesdev