neurogears/ranson-rig #44: format of parameters in TrialMetaData log file

ransona, 2021-07-27:
The parameters of the grating are grouped in what seems to be a strange way in the log file:
'{[,270,30,-60,15,],[,1,1,0,0.0667,2,0.5,],[,0,2,]}'

Is there a reason for this format? We need to then parse this at our end which is fine (we just remove all { } [ ] characters and empty spaces between commas)  but concerned that this format might change in some unpredicable way?

glopesdev, 2021-07-27:
This format follows how exactly the OSC messages are formatted between Bonsai and MATLAB/Python. It won't change unless the communication protocol changes which would also break interoperability between the rig and the clients.

Adopting a 3rd format would mean we would need to maintain that as well.

ransona, 2021-07-27:
ok! out of curiosity is there some logic to this grouping!?

glopesdev, 2021-07-29:
For future reference, the grouping is organized in `positioning`, `attributes` and `timing` parameters, and the specifics of each value can be found in:

https://github.com/neurogears/ranson-rig/blob/master/src/python/rigstim.py#L55-L74

There is a current limitation on Bonsai OSC messages that each parameter group can contain at most 8 values, but in the future we can revisit this and re-organise.

closed 2021-07-29 by glopesdev