neurogears/ranson-rig #58: timing pulses can start high or low

ransona, 2024-12-03:
timing pulses at experiment start seem to be randomly either high or low and first pulse duration is not reliable

ransona, 2024-12-09:
more generally timing pulses do not seem to be output in a steady and evenly spaced manner

glopesdev, 2025-01-27:
All examples below assume you are using [uv](https://docs.astral.sh/uv/) to manage python dependencies.

### Install harp-python

```
uv pip install harp-python
```

The photodiode synchronization pulses follow the update / render loop of the simulation. The raw photodiode signal and the simulation photodiode state updates are recorded in separate files.

### `*_FrameEvents.csv`

This file stores the logical state updates with the format:

| Column Name | Description |
|--------|--------|
| Frame | The sequential logical number of the frame |
| Timestamp | The time in seconds accumulated from the beginning |
| SyncQuad | The state of the sync quad at the specified frame update |
| Trial | The trial number |

 The change in the state of the sync quad is not done every frame, but following a pseudo-random sequence pulled from a seeded geometric distribution. The code to generate the sync sequence is at [src/workflows/Extensions/SyncSequence.cs](https://github.com/neurogears/ranson-rig/blob/d6ee25fb12495fed6a4c9d0a56d68b61d8d2d04c/src/workflows/Extensions/SyncSequence.cs).

Each time we sample a different number of frames to wait until the next flip from the distribution, and then wait for those frames to elapse with a counter. Each time the counter hits zero, we sample a new number of frames to wait.

It is possible to make this pseudo-random sequence periodic by setting the `P` parameter to 1. Interestingly, this is what we seem to have done, so in fact each logical swap is always exactly `32` frames apart (or ~270 milliseconds).

Below is a small example to read the file and get the logical frame swaps using [pandas](https://pandas.pydata.org/docs/#).

```python
frame_events_fname = r"YBBT001\2025-01-16_01_YBBT001\2025-01-16_01_YBBT001_FrameEvents.csv"
frame_events = pd.read_csv(frame_events_fname, header=0, names=['Frame', 'Timestamp', 'SyncQuad', 'Trial'])
frame_swaps = frame_events[frame_events.SyncQuad.diff().abs() > 0]
```

### `*_Behavior_Event44.bin`

This file stores the raw photodiode voltage values, harp-timestamped and sampled at 1 kHz. Even with slow drift of the coupling between the photodiode and the screen, values above 500 should reliably represent the ON quad, and conversely OFF for values below or equal 500.

Below is a small example to read the file using the [harp-python](https://github.com/harp-tech/harp-python) library.

```python
import harp
photodiode_fname = r"YBBT001\2025-01-16_01_YBBT001\2025-01-16_01_YBBT001_Behavior_Event44.bin"
photodiode = harp.read(photodiode_fname, 44, dtype=np.int16, length=2)
photodiode_swaps = photodiode[(photodiode[0] > 500).astype(np.int8).diff() != 0]
```

### Data comparison

Opening the data from a single session, we can see that we have 4512 logical swaps and 4514 photodiode swaps.

It is probably not ideal the numbers do not align outright, but because of the large jump at the start of the trial, we can actually discard all data following this interval (~800 milliseconds in this case).

If we do this, we obtain exactly 4512 swaps on each dataset with very similar timing statistics.

#### Logical swaps

```python
frame_swaps.Timestamp.diff().hist(bins=100)
frame_swaps.Timestamp.diff().describe()
```

[image]

#### Photodiode swaps

```python
load_swap = photodiode_swaps.index[photodiode_swaps.index.diff().argmax()+1]
photodiode_swaps[load_swap:].reset_index().Time.diff().describe()
photodiode_swaps[load_swap:].reset_index().Time.diff().hist(bins=100)
```

[image]

We can discuss soon how best to improve all these statistics, but at least in terms of implementation it seems to be good news that all the required building blocks seem to be in place.

ransona, 2025-01-28:
The question is to know how to use these bits of data to know when the trial starts on an independent DAQ - i.e. in order to know how to align the imaging data to trial onset for example. The independent DAQ is acquiring the photodiode signal and the SyncQuad signal. Forget for now the time when the screen actually shows a frame of the trial, we just want to know when Bonsai's time zero is.

We have the *_FrameEvents.csv file which in the 4th column has the trial number and the 2nd column has the Timestamp.  I assume we need to use this data to get a precise 'intended' trial start time?

Some questions related to this:
This Timestamp value is the value that abruptly loses time relative to an external clock, typically around the time of trial onset. Is this expected?
The trial number increment can only happen when there is a frame event right? Is the *_FrameEvents.csv file timestamp of this trial number increment the time when everything else that should happen in the trial is measured relative to or if not what is time zero of the trial?
We have 3 time bases: 1) the timestamps in the *_FrameEvents.csv (this timescale has gaps in it relative to an external clock), 2) the Harp's clock, and 3) the external DAQ clock. I am assuming the idea is to relate these by having the SyncQuad acquired on all 3. But given that one of these time bases abruptly goes out of sync this is not a matter of something straight forward.
In short, I need you to outline the steps of the algorithm to know based on the above data when the trial time zero is on an external DAQ. This does not need to be when the frame of the stimulus actually comes on the screen (this is another issue), but when the frame would ideally have come on the screen (if the monitor refresh could be synced to trial onset).

We will then need to see what can be done about when the frame actually comes on the screen and which frame of the stimulus this is (i.e. how many have been dropped at trial start as discussed last session).

glopesdev, 2025-01-28:
> This Timestamp value is the value that abruptly loses time relative to an external clock, typically around the time of trial onset. Is this expected?

@ransona If you are loading large assets at the start of each trial, then yes this would be expected, since if the delay is too long the update loop may give up trying to catch up to the dropped frames.

However, this can also give you a chance to identify the frames corresponding to new trials, since they will be obviously much longer than any other deviation.

Alternatively, you can switch acquisition to a true pseudo-random sequence by setting `P` to a value lower than 1 and then we should be able to uniquely tag each swap in the sequence. If you want me to try to reconstruct what this would look like, change the `P` for a new test run and send me the resulting dataset.

> The trial number increment can only happen when there is a frame event right?

That is correct. Any update to the screen state can only happen by definition on a frame event, since this logs all the updates that ever happened in the update / render loop.

> I am assuming the idea is to relate these by having the SyncQuad acquired on all 3. But given that one of these time bases abruptly goes out of sync this is not a matter of something straight forward.

> In short, I need you to outline the steps of the algorithm to know based on the above data when the trial time zero is on an external DAQ.

If the Harp and the DAQ are seeing the same stream of photodiode data at all times, then the steps described above should be identical to recover each sync pulse timing on the external DAQ.

This should be true as long as the DAQ is initialized before any quad shows up in the system.

> This does not need to be when the frame of the stimulus actually comes on the screen (this is another issue)

What the above algorithm gives you is the time when the sync quad actually flipped on the screen. How this relates to the timing of each individual stimulus will then depend on the trial-by-trial parameters of each stimulus.

ransona, 2025-01-28:
I still do not see how to know in external DAQ time when the trial _should_ have started - i.e. when Bonvision resets the trial timer to zero. I have spent many days trying to use what I think is the approach you are describing but I do not think it works.

I would like to try to write down the steps:

1. Identify in Harp time the time of each photodiode signal swap (from *_Behavior_Event44.bin) **DONE**
2. Identify in External DAQ time the time of each photodiode signal swap (from our DAQ output) **DONE**
3. Identify in Bonsai 'render loop time' the time of photodiode signal swaps (from *_FrameEvents.csv column 2) **DONE**
4. Use times of photodiode signal swaps on each system to make a lookup function, and use linear interpolation to infer time points between photodiode pulses **DONE**
5. Identify in Bonsai 'render loop time' the time of trial number increments (from *_FrameEvents.csv column 4) **DONE**
6. Convert these using lookup function to external DAQ time **DONE**

Does this sound right? Is the 'render loop time' of a trial number increment the time when Bonsai resets the trial time to zero? Or does this happen sometime before in parallel to the render loop? This reset happening in step with the render loop seems to be a prerequisite for using trial number increments to know when trial time zero is. Am I understanding correctly?

Would there not be a problem when the trial number increment happens away from the time of the photodiode signal swap because the lookup function is then no longer accurate?

I hope you can see the apparent gaps here?

glopesdev, 2025-01-29:
@ransona That dataset is indeed more representative and there seems to be a consistent gap at the beginning of each trial. We can plot the distribution of time deltas across all logical frame swaps to see this:

```python
frame_swaps = frame_swaps.assign(Delta=frame_swaps.Timestamp.diff())
frame_swaps.boxplot('Delta')
```

[image]

If we count the number of swap events having deltas located in the upper distribution (`Delta > 0.53`), we find 119 swap events, which matches the number of trials (120, you have to add +1 to account for the fact the first logical swap is already in trial 0).

If we now count the number of physical photodiode swap events with the same time delta, we find 120 swaps, which also matches the number of trials (here we don't need to add + 1 since the DAQ / photodiode is started before the first logical swap).

Pairing these events as the start of each trial on a trial-by-trial basis should result in a more robust alignment metric.

Nevertheless, I agree it is not entirely satisfactory so if we really cannot optimize the video startup we might want to switch to the peudo-random method altogether and set `P` to a value smaller than 1 so we can tag and align individual frame swap sequences across the entire session.

glopesdev, 2025-01-29:
> if we really cannot optimize the video startup

@ransona To try and explore this, can you share the configuration script you used to configure the trials for the `YBBT002` dataset?

ransona, 2025-01-29:
for the record here is a plot showing how the times of timing pulses show up on the various different data acquisition systems (relative to bonsai timestamp time). DAQ refers to an external DAQ acquiring both the digital and photodiode timing signals.

[image]

closed 2025-03-19 by glopesdev