neurogears/ranson-rig #70: bug in trial meta data is being stored in *_TrialMetadata.csv - 0.5 becomes 0,5

ransona, 2025-03-06:
Something is now wrong in the way the trial meta data is being stored in *_TrialMetadata.csv. This is one trial's data with the new workflow:
{'{[,0,30,30,-20,-20,],[,1,1,0,0,05,2,**0,5**,],[,0,1,]}'}

And this is with the old workflow:
 "{[,0,30,30,-20,-20,],[,1,1,0,0.05,2,**0.5**,],[,0,1,]}"

You see how the 0.05 has been stored as '0,05,'?

glopesdev, 2025-03-18:
@ransona This actually turns out to be a bug in the OSC package (https://github.com/bonsai-rx/bonsai/issues/2184). Most likely the reason we didn't spot it at the time is the computers were still configured using UK settings. Sadly under Latin languages such as PT or ES, the decimal separator is `,` rather than `.`, and because of this issue the culture of the OS installed in the computer will be preferred.

I will fix it for the next release of the package, but in the meantime I will probably write a custom logging operator to make sure this is dealt with correctly in the rig.

glopesdev, 2025-03-21:
@ransona yes, that is correct, let me know if that doesn't work and we can try to override the formatting behaviour from a custom script.