neurogears/ranson-rig #71: not all harp data is logged

ransona, 2025-03-06:
Specifically the digital input and output used for lick detection and reward triggering are absent.

glopesdev, 2025-03-18:
@ransona from what I can see in the code, all hardware events should already be recorded in the harp binary data logs. As long as the correlation between the photodiode signal and the harp clock is working, you can use it to project any harp time into screen time, and vice-versa.

In the task logic code, a reward will trigger a solenoid valve pulse, which is also logged and timestamped in hardware. I believe at the time we agreed that for behavior analysis it was much more relevant to log the time at which the valve exactly opens, and conversely the time at which the lick is exactly detected, rather than logical events in the simulation loop (since there is an inevitable delay between these and the physical hardware action).

The only responses we log in simulation time I believe is the outcome of passive trials (and possibly the end of the VR corridor), since for those there is indeed no hardware analogue rather than the update of the screen state itself.

Maybe in general we should go into more detail about how exactly to access and correlate all Harp hardware signals so we can make sure all binary files are accessible? At the time we didn't have [harp-python](https://github.com/harp-tech/harp-python) yet, but with this library it should be much easier to dive into these files directly.

glopesdev, 2025-03-18:
On a tangent, I do see in the code that the lick detection responses are actually also being logged in screen time, and should be stored in the file `*_Responses.csv`.

This was probably done for replay trials, as we need to keep a record of all recorded inputs in logical screen update time, but for analysis I still feel the actual lick detection events in harp time are the correct events to use, e.g. to correlate with neural activity.